gary-rowe/hid4java #109: write()和sendFeatureReport()方法都报错-1

a751116023, 2020-12-16:
厂家软件监听到的USB用的是这个函数<Bulk or Interrupt Transfer>，我用hid4java的write()和sendFeatureReport()方法都报错-1，监听到的数据包用的还是<Get Descriptor Request>函数。
比如我的代码，设备已经被打开：
```
        HidServices hidServices = HidManager.getHidServices();
        HidDevice hidDevice = hidServices.getHidDevice(0xFFFE, 0x0091, "0690460036");
        boolean open = hidDevice.open();

        byte[] buff2 = new byte[7];
        buff2[0] = (byte) 0xFA;
        buff2[1] = (byte) 0x06;
        buff2[2] = (byte) 0xFF;
        buff2[3] = (byte) 0x40;
        buff2[4] = (byte) 0x04;
        buff2[5] = (byte) 0x0A;
        buff2[6] = (byte) 0x8E;

        int i = hidDevice.write(buff2, buff2.length, (byte) 7);
        String lastErrorMessage1 = hidDevice.getLastErrorMessage();
        int i1 = hidDevice.sendFeatureReport(buff2, (byte) buff2.length);
        String lastErrorMessage2 = hidDevice.getLastErrorMessage();
```
[image]

a751116023, 2020-12-16:
这是用hid4java的方法给读卡器发送数据时，监听到的数据
[image]

gary-rowe, 2020-12-16:
Thank you for bringing this to my attention. I need more details before I can comment. 

Which version of `hid4java` are you using? 

What are you expecting to see that you are not? 

An error code of -1 indicates a failure so can you provide more details.

Google translate:
谢谢您引起我的注意。 我需要更多详细信息才能发表评论。

您使用的是哪个版本的hid4java？

您期望看到什么不是您？

错误代码-1表示失败，因此您可以提供更多详细信息。

a751116023, 2020-12-16:
> 谢谢您引起我的注意。我需要更多详细信息才能发表评论。
> 
> `hid4java`您使用的是哪个版本？
> 
> 您期望看到什么不是您？
> 
> 错误代码-1表示失败，因此您可以提供更多详细信息。
> 
> Google翻译：
> 谢谢您引起我的注意。我需要更多详细信息才能发表评论。
> 
> 您使用的是哪个版本的hid4java？
> 
> 您期望看到什么不是您？
> 
> 错误代码-1表示失败，因此您可以提供更多详细信息。

我用的0.7.0版本的

a751116023, 2020-12-16:
> Thank you for bringing this to my attention. I need more details before I can comment.
>
> Which version of `hid4java` are you using?
>
> What are you expecting to see that you are not?
>
> An error code [...]

这个是监听厂家软件发送的usb数据包的截图:
[image]

这个是hid4api发送的write()函数的数据包的截图:
[image]

a751116023, 2020-12-17:
> Thank you for bringing this to my attention. I need more details before I can comment.
>
> Which version of `hid4java` are you using?
>
> What are you expecting to see that you are not?
>
> An error code of [...]

问题解决了，原因是一个读卡器在操作系统注册了2个HID device，hid4java默认拿的第一个设备操作，第二个是可以正常使用的。我改下代码重新打包下使用

closed 2020-12-17 by a751116023

gary-rowe, 2020-12-17:
That is good news. I'm glad you were able to fix the problem. Good luck with your project.

Google translate:
这是个好消息。 很高兴您能够解决此问题。 祝您的项目好运。