gary-rowe/hid4java #110: 为什么每次read()只能读取49个byte的数据，我传过去256长度的byte[]数组，还是只能读取到49个byte？

a751116023, 2020-12-19:
为什么每次read()只能读取49个byte的数据，我传过去256长度的byte[]数组，还是只能读取到49个byte？
HID设备每次返回128个byte的数据，但是我只能接收到48个byte，造成接收数据混乱。。请问有方法避免这个问题吗？

gary-rowe, 2020-12-19:
Google translate:
Why can only read 49 bytes of data every time read(), I pass the 256-length byte[] array, or can only read 49 bytes? The HID device returns 128 bytes of data each time, but I can only receive 48 bytes, causing confusion in the received data. Is there a way to avoid this problem?

gary-rowe, 2020-12-19:
Just using read() will use defaults of 64 bytes and a 100ms timeout. You might want to consider using read(128,200) for a more flexible approach.

In v0.8.0 (not released yet but available on the develop branch) there is a readAll() method that will read all bytes coming from the device in a single hit.

Google translate:
仅使用read（）将使用64字节的默认值和100ms的超时时间。 您可能需要考虑使用read（128,200）获得更灵活的方法。

在v0.8.0（尚未发布，但可在develop分支上使用）中，有一个readAll（）方法将在一次命中中读取来自设备的所有字节。

a751116023, 2020-12-19:
hid4java-0.7.0使用的hidapi是0.7还是0.8？

gary-rowe, 2020-12-19:
Please translate your questions to English using [Google Translate](https://translate.google.com). 

Google translate:
请使用[Google翻译]（https://translate.google.com）将您的问题翻译成英语。

gary-rowe, 2020-12-19:
`hid4java-0.7.0` uses `hidapi-0.9.0`. The release policy is to use the latest available release of `hidapi` at the time.

a751116023, 2020-12-20:
English:
According to the problems I encountered in actual use, more than one Hiddevice was found by SN using the same VID and PID.
I've added a getHidDeviceList() method to the hidservices class that returns all of the List<hiddevice> criteria.
Then test the return value using the hiddevice.write () method, which if > 0 is the operable target device

Chinese:
根据我在实际使用中遇到的问题，使用相同的vid和pid, sn查找到的hiddevice不止一个。
我在hidservices类增加了getHidDeviceList()方法，该方法返回所有满足条件的List<hiddevice>。
然后使用hidDevice.write()方法测试返回值,如果> 0,该hiddevice为可操作的目标设备

closed 2020-12-23 by a751116023