gary-rowe/hid4java #119: Linux build linux-armel is failing

tresf, 2021-08-06:
Running `./build-hidapi.sh linux-armel` results in an error.
* When running this build an error occurs stating that claims a header is missing.
   ```
   hid.c:44:10: fatal error: libudev.h: No such file or directory
   44 | #include <libudev.h>
      |          ^~~~~~~~~~~
   ```

Furthemore, the toolchain seems to be wrong:
* The toolchain seems to be incorrectly using the `-armhf` suffix, it should just be `-arm`
   * `gcc-arm-linux-gnueabihf` should be `gcc-arm-linux-gnueabi`
   * `--host=arm-linux-gnueabihf` should be `--host=arm-linux-gnueabi`
   * `libudev-dev:armhf` should be `libudev-dev:armel`

I attempted to fix this by hand, but after a few unsuccessful builds dockcross is telling me `You don't have enough free space in /var/cache/apt/archives/.`, which I can't figure out how to fix. 😆

gary-rowe, 2022-12-05:
As an update I've taken a look at this problem recently and the problem remains. There seems to be no good way to get the build to install `/usr/includes/libudev.h`. I'll leave this issue open

tresf, 2023-11-06:
> As an update I've taken a look at this problem recently and the problem remains. There seems to be no good way to get the build to install `/usr/includes/libudev.h`.
> 
> I'll leave this issue open as it's presently causing the `linux-armel` support to be missing in the current `develop` branch.

Had a near-identical issue with RISC-V.   The workaround was to simply copy this header into the work directory.

https://github.com/gary-rowe/hid4java/blob/c65bfd1a5250305b93dd165fc6e3bc027649c563/build-hidapi.sh#L317

Now, I can't prove this theory, but I believe the cross-compilers are deliberately avoiding importing headers from the `/usr/include` due some some `sysroot` assumptions.   The reason I believe this is because on the RISC-V branch, I had explicitly added `-I/usr/include` to the `INCLUDES` automake variable and saw it with `make V=1`, but still received the same error.

gary-rowe, 2024-03-29:
Looks like recent PRs have fixed this (see #143)

closed 2024-03-29 by gary-rowe