gary-rowe/hid4java #120: Building for MacOS should encourage setting an SDK version

tresf, 2021-08-07:
HID4Java should be built using an older SDK, such as 10.9.

### MacOS x86_64
* At time of writing this, the 10.9 SDK is the oldest SDK which Apple will allow notarization for. 
* Normally the SDK can be [extracted from a XCode 6.2 (apple.com)](http://developer.apple.com/devcenter/download.action?path=/Developer_Tools/Xcode_6.2/Xcode_6.2.dmg).
* Alternately the SDK can be obtained as a stand-alone archive from a mirror such as: https://github.com/phracker/MacOSX-SDKs/releases/download/11.3/MacOSX10.9.sdk.tar.xz (much smaller file size)

### MacOS aarch64
* At the time of writing this, the 11.0 SDK is the oldest SDK which is capable of building for the Apple Silicon/M1 aarch64 CPU.
* Normally the SDK can be [extracted from a XCode 12.2 (apple.com)](https://developer.apple.com/services-account/download?path=/Developer_Tools/Xcode_12.2/Xcode_12.2.xip)
* Alternately the SDK can be obtained as a stand-alone archive from a mirror such as: https://github.com/phracker/MacOSX-SDKs/releases/download/11.3/MacOSX11.0.sdk.tar.xz (much smaller file size)

Configuring the environment must be done in two steps:
* Specifying the full path to the SDK
   ```bash
   # e.g.
   export SDKROOT=/Library/Developer/CommandLineTools/SDKs/MacOSX10.9.sdk 
   ```
* Specifying the target OS
   ```bash
   # e.g.
   export MACOSX_DEPLOYMENT_TARGET=10.9
   ```
* To confirm, you can check the outputted file using `otool -l`
   ```bash
   # e.g.
   otool -l src/main/resources/darwin-x86-64/libhidapi.dylib
   # or
   otool -l src/main/resources/darwin-x86-64/libhidapi.dylib |grep minos
   ```

gary-rowe, 2024-03-29:
Is this still an issue?

tresf, 2024-03-29:
> Is this still an issue?

It was merged here
* https://github.com/gary-rowe/hid4java/pull/

... However, I believe it's missing on the develop branch.

tresf, 2024-04-15:
@gary-rowe which branch is this on?  Or did you decide not to add it?

Here's an example of how this was done on another branch, but the code was never merged:

https://github.com/gary-rowe/hid4java/blob/e40bdfe0b71b15b8988f5e0bc22c952a3faa0653/build-hidapi.sh#L330-L334

gary-rowe, 2024-04-16:
Hi @tresf . Looks like it was missed with all the other changes to the `build-hidapi.sh` script. I've just pushed some changes to `develop` which should address the minimum targets on macOS. I've tested the new new `.dylibs` on a couple of macOS machines and they seem OK but if you'd like to check them over I'd appreciate it.

tresf, 2024-04-16:
@gary-rowe thanks for the attention to detail however I think there may be some overkill items, such as restarting `zsh` with another architecture.

https://github.com/gary-rowe/hid4java/blob/672f0a7adaeac0ae9ead2cca32548ed0974886c6/build-hidapi.sh#L430

... in my experience, this was just fine by passing the arch flag to the build system.

https://github.com/gary-rowe/hid4java/blob/e40bdfe0b71b15b8988f5e0bc22c952a3faa0653/build-hidapi.sh#L337

I think this approach is cleaner and less confusing.

PR #116 (2022-12-05): Improvements to MacOS builds
Note before merging: This PR contains commits from #115, but they can be refactored out once that branch is merged into `develop`.

Relevant commits:
* 1e2be9a - Add darwin-aarch64 to dockcross
* fba8f73 - Bump JNA to 5.8.0 (allows separate `.dylib` files for separate architectures on MacOS)
* 88c66e9 - Switch to JNA 5.8.0 darwin suffix
* 88c66e9 - Gracefully handle missing folders, files
* 408420c - Remove problematic calls to `make clean` (Closes #122)
* e40bdfe - Add hints for `SDKROOT`, `MACOSX_DEPLOYMENT_TARGET` (Closes #120)

Related #108
commit ef4c5f74 (2022-12-05): Improvements to MacOS builds (#116)

* #93 Create placeholder for hidapi win32-aarch64

* #93 Add POM version for local JNA issue 93 support

* #93 Add POM version for JNA 5.7.0-SNAPSHOT

* #93 Add local library reference for faster JNA turnaround

* #93 Re-instate missing DLL and reference JNA-5.7.0-SNAPSHOT

* Fix buld-hidapi.sh
Closes #114
Make colors easier to read in the shell script

* Add win32-aarch64 to dockcross

* Fix JNA dependency

* Fix unit test
Wait 30s for testing as logs indicate

* Add darwin-aarch64 to dockcross

* Bump JNA to 5.8.0

* Switch darwin to new jna suffix
Also a [... cut]