gary-rowe/hid4java #125: Initial attach events (with setAutoStart(false)) not generated after HidServices.stop() + HidServices.start()

MichaelEnke, 2021-12-01:
Not sure if to categorize this as error report or feature request:
When we set setAutoStart(false), add a HidServicesListener and than call HidServices.start(),
for all attached devices hidDeviceAttached(HidServicesEvent) is called.
If we now call stop() and start() again, there is no hidDeviceAttached(HidServicesEvent) called.
2 Reasons for that:
1: In the stopScanThread() method, it should be checked after calling interrupt() that the tread is no longer alive, e.g. by checking isAlive() in a while loop with 10 ms sleep (it is still a few milliseconds alive), otherwise when calling stop() and immediately start(), the isScanning() method still returns true.
2: in stop() method it should be called: this.attachedDevices.clear(); in order to emit the attached event in scan() method

osaman88, 2021-12-06:
Also observed this and agree this would be a highly desirable feature/bug-fix. 

I'd like to note that clearing of HidManager.hidServices.HidDeviceManager.attachedDevices (which Michael refers to above in 2nd reason) is impossible in the current implementation without physically detaching devices. HidManager.hidServices is a static private member with no setter methods available. Once the attachedDevices map is populated, it persists around and stops alerting on already-attached devices, even if we shutdown HidServices and attempt a new start with different settings. While I was able to work around this through reflection and modifying the access permission to set HidManager.hidServices to null, this is not code anyone would want to add to production code. If there is no desire to change the existing behavior of HidServices.{start,stop} to not break legacy usage, it would be very helpful to get a method to clear HidManager.hidServices.HidDeviceManager.attachedDevices and we can call such a method ourselves ahead of HidServices.start, or alternatively, when shutting down and restarting HidServices.

gary-rowe, 2024-03-29:
Have (finally) gotten around to adding this to the `develop` branch.

closed 2024-04-14 by gary-rowe