gary-rowe/hid4java #126: HID write always fails with 64bit OS running under Raspberry PI 4

mercierm, 2021-12-03:
**Describe the bug**
The HID device attaches and opens successfully.
But the following code always fails and returns -1:

private static final byte[] message;
static {
   message = new byte[] { 0x10, 0x00, 0x00, 0x00, 0x00, 0x00, (byte) 0xff, (byte) 0x80,
                                          0x00, 0x67, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
                                        };
} 
int bytesWritten = hidDevice.write( message, 16, 0);

**To Reproduce**
Steps to reproduce the behavior:
1. Fails under 32 & 64 bit Raspbian OS on Raspberry PI 4B.
2. Runs OK under 32 bit Raspbian OS on Raspberry PI 3B

**Expected behavior**
The number of bytes send successfully (17) should be returned.

gary-rowe, 2022-12-05:
Can you try this code on your devices with the latest build from `develop` - it may have resolved the issue.

gary-rowe, 2024-03-29:
Closing due to inactivity

closed 2024-03-29 by gary-rowe