gary-rowe/hid4java #127: open has no effect - with setAutoDataRead(true) I get java.lang.IllegalStateException: Device has not been opened

Sciss, 2022-01-07:
I'm trying to open an HID using the latest (snapshot) version of hid4java. I don't know if this is a duplicate of #113 ; but in this case I don't get any error attempting to call `.open()`. Instead, when the library tries to start polling data, this happens:

```
Exception in thread "main" java.lang.IllegalStateException: Device has not been opened
	at org.hid4java.HidDevice.readAll(HidDevice.java:473)
	at org.hid4java.HidDevice.dataRead(HidDevice.java:193)
	at org.hid4java.HidDevice.startDataReadThread(HidDevice.java:121)
	at org.hid4java.HidDevice.open(HidDevice.java:300)
	at de.sciss.rogues.HIDTest$.run(HIDTest.scala:51)
```

I'm using Java 11 on Debian 11:

```
$ uname -a
Linux aleph 5.10.0-10-amd64 #1 SMP Debian 5.10.84-1 (2021-12-08) x86_64 GNU/Linux
```

This is the output of `dmesg` when the device is connected:

```
[768929.931938] usb 3-1: new low-speed USB device number 26 using xhci_hcd
[768930.084589] usb 3-1: New USB device found, idVendor=0079, idProduct=0006, bcdDevice= 1.07
[768930.084593] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[768930.084595] usb 3-1: Product: Generic   USB  Joystick  
[768930.084596] usb 3-1: Manufacturer: DragonRise Inc.  
[768930.093749] input: DragonRise Inc.   Generic   USB  Joystick   as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/0003:0079:0006.001E/input/input58
[768930.093805] dragonrise 0003:0079:0006.001E: input,hidraw3: USB HID v1.10 Joystick [DragonRise Inc.   Generic   USB  Joystick  ] on usb-0000:00:14.0-1/input0
[768930.093809] dragonrise 0003:0079:0006.001E: Force Feedback for DragonRise Inc. game controllers by Richard Walmsley <richwalm@gmail.com>
```

It's a DYI game controller circuit board. I also tried with an external mouse, same problem. Here is code, which is essentially the example from the hid4java repository:

```scala
import org.hid4java.{HidManager, HidServices, HidServicesListener, HidServicesSpecification}
import org.hid4java.event.HidServicesEvent
import org.hid4java.jna.HidApi

// more or less verbatim translation from hid4java by Gary Rowe (MIT License)
object HIDTest:
  def main(args: Array[String]): Unit =
    val hidServicesSpecification = new HidServicesSpecification
    hidServicesSpecification.setAutoDataRead(true)
    hidServicesSpecification.setDataReadInterval(500)

    val hidServices = HidManager.getHidServices(hidServicesSpecification)
    hidServices.addHidServicesListener(Listener)
    
    println("Manually starting HID services.")
    hidServices.start()
    
    println("Enumerating attached devices...")
    
    // Provide a list of attached devices    
    import scala.jdk.CollectionConverters._
    val hidDevices = hidServices.getAttachedHidDevices.asScala
    for hidDevice <- hidDevices do
      println(hidDevice)

    if hidDevices.nonEmpty then
      val hidDevice = hidDevices.last
      println(s"Opening for four seconds... isClosed? ${hidDevice.isClosed}")
      hidDevice.open()
      Thread.sleep(4000)
      hidDevice.close()

    Thread.sleep(1000)

    println("Triggering shutdown...")
    hidServices.shutdown()

  object Listener extends HidServicesListener:
    override def hidDeviceAttached(event: HidServicesEvent): Unit =
      println("Device attached: " + event)

    override def hidDeviceDetached(event: HidServicesEvent): Unit =
      println("Device detached: " + event)
  
    override def hidFailure(event: HidServicesEvent): Unit =
      println("HID failure: " + event)
  
    override def hidDataReceived(event: HidServicesEvent): Unit =
      println("Data received.")

  end Listener
```

I also tried `HidApi.useLibUsbVariant = true` but it doesn't change anything.

Sciss, 2022-01-07:
I checked with 'purejavahidapi', this works in the same situation, so the controller is definitely correctly readable. Perhaps I'm just missing a way to trigger the reading?

gary-rowe, 2022-12-05:
It's been some time since this occurred. Is it still an issue?

closed 2024-03-29 by gary-rowe