gary-rowe/hid4java #128: Add module name

satsen, 2022-02-13:
Please add a module name to the library. It should probably be `org.hid4java`.

gary-rowe, 2022-12-04:
Thanks for the suggestion, but at present `hid4java is keeping with Java 8 for source and target output. This is due to many developers sticking with the Long Term Support variants of Java. Currently Java 8 is the lowest (with 11, 17 being other staging posts) and to reach the widest audience I'm holding back.

However, if you think this will significantly improve the library's reach let me know, and perhaps offer a suitable pull request which I'd be happy to review and merge if convinced.

satsen, 2022-12-08:
Hello, I disagree with your argument for supporting Java 8 because if one were to think like that, then there's extended support from Oracle until 2030 and it would mean libraries should never move on (even latest Spring is going for Java 17).

It is your choice though and I'm not gonna discuss it, but can you at least add an Automatic-Module-Name to the manifest for now? It works as well and is much better than no name. Thank you.

satsen, 2022-12-08:
_(Also, support for Java 8 ended in March of 2022. What is left is extended support and that's probably really expensive and nobody should still be using it.)_

closed 2023-03-01 by satsen