gary-rowe/hid4java #129: I am not scanning my device using UsbHidEnumerationExample. Do I need to do anything to set the MAC?

RKaixuan, 2022-07-07:
I am not scanning my device using UsbHidEnumerationExample. Do I need to do anything to set the MAC?

junction2412, 2022-07-07:
Well it worked fine for me in the underneath code.

    import org.hid4java.*
    import java.nio.ByteBuffer
    import java.nio.file.Files
    import java.nio.file.Paths
    import java.nio.file.StandardOpenOption

    fun main(args: Array<String>) {
        val path = Paths.get(System.getenv("USERPROFILE"), "Desktop", "hid", "test.txt")
        if (!Files.exists(path))
            Files.createFile(path)

        val arr: Array<HidDevice?> = Array(1) { null }

        for (device: HidDevice in HidManager.getHidServices().attachedHidDevices) {
            println("manufacturer: ${device.manufacturer}, product: ${device.product}, usage_page: ${device.usagePage}, usage: ${device.usage}")

            if (device.usagePage == 0x01 && device.usage == 0x05)
                arr[0] = device
        }

        if (arr[0] != null) {
            println("gamepad found\n")
            val gamepad: HidDevice = arr[0]!!
            gamepad.open()

            val start: Long = System.currentTimeMillis()

            do {
                val read: Array<Byte> = gamepad.read()!!
                val array = ByteArray(read.size)

                for (index in read.indices)
                    array[index] = read[index]

                for (i in 0..(array.size / 8))
                    print(String.format("%x ", ByteBuffer.wrap(array.copyOfRange(i, i + 8)).long))
                println()

                Files.write(
                    path,
                    array,
                    StandardOpenOption.APPEND
                )
            } while (System.currentTimeMillis() - start <= 5_000)

            gamepad.close()
        }
    }

Gotta admit I'm very impressed by this api, yet I still have to figure out how the input-reports work to be able to do something useful with it. 😅

RKaixuan, 2022-07-12:
I have new problem, HipApi.open is returning null when I send information to the device for the second time

junction2412, 2022-07-12:
Did you close the former connection properly before opening it the 2nd time?

RKaixuan, 2022-07-12:
Thinks, I got it

RKaixuan, 2022-07-28:
Why 32-bit windows7 can't send data to the device, but can receive data

gary-rowe, 2022-12-05:
Since this is an old question/answer support issue I'll close it for now.