gary-rowe/hid4java #133: When will 0.8.0 be released as example code doesn't compile when tried with latest published jar?

garethrandall, 2022-12-02:
After adding the latest 0.7.0 jar to a project (via Maven's pom.xml), then trying to copy and paste the example Fido2AuthenticationExample.java code in to see how it works, the project fails to compile because in the 0.7.0 release, there is no method called:

org.hid4java.HidServicesListener.hidDataReceived(HidServicesEvent event)

I’ve also confirmed this in the javadoc jar:
https://repo1.maven.org/maven2/org/hid4java/hid4java/0.7.0/hid4java-0.7.0-javadoc.jar

I can see that the code has moved on since 0.7.0, so is it time for a 0.8.0 release that includes everything so far?

Thanks!

gary-rowe, 2022-12-02:
I agree it's been a long time coming. I was hoping to get this release done a lot sooner but getting access to a suitable Mac for compiling the `aarch64` library proved tricky (hardware failures and procurement). I'm looking into making time for a release in the next week.

gary-rowe, 2022-12-04:
I've pushed some code to `develop` that will be used as the basis for an 0.8.0 release shortly. I'll close this issue to avoid clutter.

closed 2022-12-04 by gary-rowe

jackrusher, 2022-12-14:
(watching like a hawk for the new release 😁)

gary-rowe, 2022-12-14:
If you want to try it out, the `develop` branch will be very close to 0.8.0. If anything doesn't work, please let me know.

jackrusher, 2022-12-14:
@gary-rowe I had a look for `develop-SNAPSHOT` on Maven, but couldn't find it. Is the only way to help test the `develop` branch to build locally atm?

gary-rowe, 2022-12-14:
Yeah, it's just here on GitHub right now. You'll need to compile and run from the source. Instructions in the [Wiki](https://github.com/gary-rowe/hid4java/wiki/How-to-set-up-a-build-environment-from-scratch). If you know Maven, it's all very standard.

jackrusher, 2022-12-16:
@gary-rowe built flawlessly with minimal fuss on my late model aarch64 mac, runs great so far :)