gary-rowe/hid4java #134: Support for Raspberry pi 4

spicalous, 2022-12-30:
**Describe the bug**
Trying to use this library on raspberry pi4, but there are no events from library
Running through troubleshooting wiki, I found some differences on my raspberry pi 4

**To Reproduce**
Running a simple unit test
```
@Slf4j
class HIDTest {

   @Test
   void hidTestAutoStartFalse(TestInfo testInfo) {
      HidServicesSpecification specification = new HidServicesSpecification();
      specification.setAutoStart(false);

      TestHidListener hidListener = new TestHidListener();
      HidServices hidServices = HidManager.getHidServices(specification);
      hidServices.addHidServicesListener(hidListener);
      hidServices.start();
      List<HidDevice> devices = hidServices.getAttachedHidDevices();
      for (HidDevice device : devices) {
         log.info("{} Startup info device=\"{}\"", testInfo.getDisplayName(), device);
      }
   }

   @Slf4j
   private static class TestHidListener implements HidServicesListener {

      @Override
      public void hidDeviceAttached(HidServicesEvent event) {
         log.info("hidDeviceAttached event={}", event);
      }

      @Override
      public void hidDeviceDetached(HidServicesEvent event) {
         log.info("hidDeviceDetached event={}", event);
      }

      @Override
      public void hidFailure(HidServicesEvent event) {
         log.info("hidFailure event={}", event);
      }
   }

}
```

**Expected behavior**
Events to be published and logged

**Additional info**
```
$ uname -a
Linux ubuntu 5.15.0-1021-raspi #23-Ubuntu SMP PREEMPT Fri Nov 25 15:27:43 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.1 LTS
Release:	22.04
Codename:	jammy
```

I noticed there is an arm section in the https://github.com/gary-rowe/hid4java/wiki/Troubleshooting#arm guide

However my path for this file is different
```
$ find . -name 'libudev.so.1'
./usr/lib/aarch64-linux-gnu/libudev.so.1
```

Should I create a symlink by tweaking the path? e.g. `sudo ln -sf /usr/lib/aarch64-linux-gnu/libudev.so.1 /lib/arm-linux-gnueabihf/libudev.so.0`?

Sorry for a complete noob question but is there somewhere further I can check logs? At the os level for example?

spicalous, 2023-02-04:
Tried both below with no luck, but I'm pretty much shooting in the dark :smiling_face_with_tear: 
```
sudo ln -sf /usr/lib/aarch64-linux-gnu/libudev.so.1 /lib/arm-linux-gnueabihf/libudev.so.0
sudo ln -sf /usr/lib/aarch64-linux-gnu/libudev.so.1 /lib/aarch64-linux-gnu/libudev.so.0
```

martinpaljak, 2023-09-27:
AFAIU there is no support for the hidapi binaries inside the distribution for aarch64 linux.

pwitiw, 2023-12-05:
@spicalous have you resolved the issue? I am facing similar situation

gary-rowe, 2024-03-29:
Closing due to inactivity

closed 2024-03-29 by gary-rowe