gary-rowe/hid4java #135: MacOS ARM incompatible

isXander, 2023-04-14:
**Describe the bug**
```
org.hid4java.HidException: Hidapi did not initialise: dlopen(/Users/mattiadoronzo/Library/Caches/JNA/temp/jna13125369681904347397.tmp, 0x0009): tried: '/Users/mattiadoronzo/Library/Caches/JNA/temp/jna13125369681904347397.tmp' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/Users/mattiadoronzo/Library/Caches/JNA/temp/jna13125369681904347397.tmp' (no such file), '/Users/mattiadoronzo/Library/Caches/JNA/temp/jna13125369681904347397.tmp' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64'))
	at org.hid4java.HidDeviceManager.<init>(HidDeviceManager.java:87) ~[org_hid4java_hid4java-0.7.0-b365afc0be3fbb51.jar:?]
	at org.hid4java.HidServices.<init>(HidServices.java:82) ~[org_hid4java_hid4java-0.7.0-b365afc0be3fbb51.jar:?]
	at org.hid4java.HidManager.getHidServices(HidManager.java:75) ~[org_hid4java_hid4java-0.7.0-b365afc0be3fbb51.jar:?]
	at dev.isxander.controlify.controller.hid.ControllerHIDService.start(ControllerHIDService.java:33) ~[Controlify-1.1.0-beta.2+1.19.4-fabric-1.19.4.jar:?]
	at dev.isxander.controlify.Controlify.preInitialiseControlify(Controlify.java:149) ~[Controlify-1.1.0-beta.2+1.19.4-fabric-1.19.4.jar:?]
	at net.minecraft.class_310.handler$zgf000$controlify$onInputInitialized(class_310.java:3015) ~[client-intermediary.jar:?]
	at net.minecraft.class_310.<init>(class_310.java:515) ~[client-intermediary.jar:?]
	at net.minecraft.client.main.Main.main(Main.java:198) ~[minecraft-1.19.4-client.jar:?]
	at net.fabricmc.loader.impl.game.minecraft.MinecraftGameProvider.launch(MinecraftGameProvider.java:462) ~[fabric-loader-0.14.18.jar:?]
	at net.fabricmc.loader.impl.launch.knot.Knot.launch(Knot.java:74) ~[fabric-loader-0.14.18.jar:?]
	at net.fabricmc.loader.impl.launch.knot.KnotClient.main(KnotClient.java:23) ~[fabric-loader-0.14.18.jar:?]
	at org.prismlauncher.launcher.impl.StandardLauncher.launch(StandardLauncher.java:88) ~[NewLaunch.jar:?]
	at org.prismlauncher.EntryPoint.listen(EntryPoint.java:126) ~[NewLaunch.jar:?]
	at org.prismlauncher.EntryPoint.main(EntryPoint.java:71) ~[NewLaunch.jar:?]
Caused by: java.lang.UnsatisfiedLinkError: dlopen(/Users/mattiadoronzo/Library/Caches/JNA/temp/jna13125369681904347397.tmp, 0x0009): tried: '/Users/mattiadoronzo/Library/Caches/JNA/temp/jna13125369681904347397.tmp' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/Users/mattiadoronzo/Library/Caches/JNA/temp/jna13125369681904347397.tmp' (no such file), '/Users/mattiadoronzo/Library/Caches/JNA/temp/jna13125369681904347397.tmp' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64'))
	at com.sun.jna.Native.open(Native Method) ~[jna-5.12.1.jar:?]
	at com.sun.jna.NativeLibrary.loadLibrary(NativeLibrary.java:283) ~[jna-5.12.1.jar:?]
	at com.sun.jna.NativeLibrary.getInstance(NativeLibrary.java:467) ~[jna-5.12.1.jar:?]
	at com.sun.jna.Library$Handler.<init>(Library.java:192) ~[jna-5.12.1.jar:?]
	at com.sun.jna.Native.load(Native.java:622) ~[jna-5.12.1.jar:?]
	at com.sun.jna.Native.load(Native.java:596) ~[jna-5.12.1.jar:?]
	at org.hid4java.jna.HidrawHidApiLibrary.<clinit>(HidrawHidApiLibrary.java:36) ~[org_hid4java_hid4java-0.7.0-b365afc0be3fbb51.jar:?]
	at org.hid4java.jna.HidApi.init(HidApi.java:105) ~[org_hid4java_hid4java-0.7.0-b365afc0be3fbb51.jar:?]
	at org.hid4java.HidDeviceManager.<init>(HidDeviceManager.java:84) ~[org_hid4java_hid4java-0.7.0-b365afc0be3fbb51.jar:?]
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: macOS 13.3

martinpaljak, 2023-09-14:
There was #108 but no releases after that. Anything we could do? Apple has transitioned fully to apple silicon by now, so this is aserious blocker...

umjammer, 2023-09-24:
i use hid4java develop-SNAPSHOT via jitpack on mac studio m2 ultra
that contains aarch64 dylib

see https://jitpack.io/#gary-rowe/hid4java

and my pom.xml like

```
  <repositories>
    <repository>
      <id>jitpack.io</id>
      <url>https://jitpack.io</url>
    </repository>
  </repositories>

   :

    <dependency>
      <groupId>com.github.gary-rowe</groupId>
      <artifactId>hid4java</artifactId>
      <version>develop-SNAPSHOT</version>
    </dependency>
```

martinpaljak, 2023-09-24:
Yep, I also resorted to a private build of the develop branch, which works fine on apple silicon and also fixed issues on Linux. But would be lovely to have a new release!

closed 2024-04-14 by gary-rowe