gary-rowe/hid4java #137: HidApi.read returning -2(DEVICE_ERROR) and "Device not initialised" seen when trying to fetch SN, PID

koushikbiz, 2023-04-21:
I am trying to open and read from a TRF7970A EVM board (by Texas Instruments) connected to my Mac. The device shows up in "ioreg -lfxp IOUSB" as a "USB to UART Bridge Controller" and gives the vid, pid and serial number.
The device path in my Mac is as follows (machine name redacted)
koushikdas@xxxx ~ % ls /dev/tty* | grep usb
/dev/tty.usbserial-0001
I changed the permission as follows:
sudo chmod -R 777 /dev/tty.usbserial-0001

Below is my code snippet:
        int vid = (int) Long.parseLong(Integer.toHexString(0x10c4), 16);
        int pid = (int) Long.parseLong(Integer.toHexString(0xea60), 16);
        try {
            HidApi.init();
//            HidDeviceStructure hidDeviceStructure = HidApi.open("/dev/tty.usbserial-0001");
            HidDeviceStructure hidDeviceStructure = HidApi.open(vid, pid, "0001");
            HidApi.setNonBlocking(hidDeviceStructure, true);
            log("Reading device data " + HidApi.read(hidDeviceStructure, new byte[64]));
            log("Device SN " + HidApi.getSerialNumber(hidDeviceStructure));
            log("Device PID " + HidApi.getProductId(hidDeviceStructure));
            log("Device Manufacturer " + HidApi.getManufacturer(hidDeviceStructure));
        } catch (Exception e) {
            log("Not able to open Hid device");
            e.printStackTrace();
        }

When I run the code, I see the following in console (package name redacted):
00:22:44.918 [ForkJoinPool-1-worker-1] INFO xxx - Reading device data -2
00:22:44.918 [ForkJoinPool-1-worker-1] INFO xxx - Device SN Device not initialised
00:22:44.918 [ForkJoinPool-1-worker-1] INFO xxx - Device PID Device not initialised
00:22:44.918 [ForkJoinPool-1-worker-1] INFO xxx - Device Manufacturer Device not initialised

I also tried setting HidApi.useLibUsbVariant = true; but got the same result.
Any suggestion to resolve this would be helpful.

gary-rowe, 2023-04-21:
Have you tried using the latest code in the `develop` branch? There have been a few updates to the `hidapi` library that haven't been merged into `master` yet.

koushikbiz, 2023-04-21:
Thanks Gary. Is there a way I can use the develop branch in my Gradle project without cloning the repo? 
I tried with below 2 approaches:
//Approach 1 - JitPack in build.gradle
repositories {
    jcenter()
    mavenLocal()
    mavenCentral()
    maven { url 'https://jitpack.io' }
}

dependencies {
...
implementation('org.hid4java:hid4java:develop-SNAPSHOT')
...
}
//Approach 2 - Gradle plugin in settings.gradle
plugins {
    id 'com.alexvasilkov.git-dependencies' version '2.0.1'
}

and the following in build.gradle:
git {
    implementation 'https://github.com/gary-rowe/hid4java.git', {
        tag 'develop'
    }
}

In both cases Gradle is not able to resolve hid4java

koushikbiz, 2023-04-26:
Hi Gary,
To get around the above problem of adding develop-SNAPSHOT as a dependency, I downloaded and copied the java files from src/main/java/org/hid4java to my project workspace. Also copied the darwin libraries to my project src/main/resources folder. Changed the package pointers in files accordingly. But I still see the same output (i.e read(...) returns -2 and Device not initialised for other details) when I run the program. Any suggestion?
[image]

gary-rowe, 2024-03-29:
Closing due to inactivity.

closed 2024-03-29 by gary-rowe