gary-rowe/hid4java #145: USB HID Set Report and Get Report

frankmeho, 2024-03-21:
**Describe the bug**
A clear and concise description of what the bug is.
Recently I developed my owned Java PC tool to send USB HID commands with hid4java-0.7.0.jar library. It works well on most of embedded systems such as Qualcomm platform. But one specific platform needs to send Set Report command first, then send another Get Report to receive the correct response. However, this specific platform can be sent Set Report correctly, but failed to send the Get Report, which means when I send Get Report, there's no any response. Use USB analyzer to capture USB packets, only the Set Report can be seen, not any Get Report packet was sent out.

**To Reproduce**
Steps to reproduce the behavior:
1. hidDevice.write(data, len, (byte)ReportID) to send USB HID command
2. hidDevice.read(data, 500) to receive the response data.
3. For the specific platform, I send hidDevice.write(data, len, (byte)SetReportID), and then hidDevice.write(data, len, (byte)GetReportID), it's failed to send Get Report, its returned code is wrong.

**Expected behavior**
How can I let the specific platform to work with my owned Java PC tool? Use another C/C++ SimpleHIDWrite to Set Report, and then Get Report, both can work.

**Screenshots**
If applicable, add screenshots to help explain your problem.
None

**Desktop (please complete the following information):**
Windows 10

**Smartphone (please complete the following information):**
None

**Additional context**
None

frankmeho, 2024-03-21:
@gary-rowe 
Hi, how are you?

I also tried the hid4java-develop which was uploaded to the GitHub.
I got the same result for the Get Report issue.

frankmeho, 2024-03-21:
@gary-rowe 
For Maven Project, we can only find the hid4java-0.7.0 for dependency, so it's not really replaced with hid4java-develop, what the exact problem do you think? Thanks.

	<dependency>
	<groupId>org.hid4java</groupId>
	<artifactId>hid4java</artifactId>
	<version>0.7.0</version></dependency>
        <dependency>

frankmeho, 2024-03-26:
@gary-rowe 
Gradually found that we use hid4java-develop, got the following issue of hidapi.dll.
org.hid4java.HidException: Hidapi did not initialise: Unable to load library 'hidapi':
The specified module could not be found.

The specified module could not be found.

The specified module could not be found.

Native library (win32-x86-64/hidapi.dll) not found in resource path (D:\COG_Java_USBHID_Developer\usbhid.comms\target\classes;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\commons-codec-1.16.0.jar;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\commons-collections4-4.4.jar;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\commons-io-2.15.0.jar;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\commons-math3-3.6.1.jar;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\poi-5.2.5.jar;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\poi-ooxml-5.2.5.jar;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\hid4java-0.8.0.jar;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\log4j-core-2.21.1.jar;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\xmlbeans-5.2.0.jar;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\log4j-api-2.21.1.jar;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\swt.jar;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\jna-5.13.0.jar;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\jna-platform-5.13.0.jar;C:\Users\HO006043\.m2\repository\org\apache\poi\poi\5.2.5\poi-5.2.5.jar;C:\Users\HO006043\.m2\repository\commons-codec\commons-codec\1.16.0\commons-codec-1.16.0.jar;C:\Users\HO006043\.m2\repository\org\apache\commons\commons-collections4\4.4\commons-collections4-4.4.jar;C:\Users\HO006043\.m2\repository\org\apache\commons\commons-math3\3.6.1\commons-math3-3.6.1.jar;C:\Users\HO006043\.m2\repository\commons-io\commons-io\2.15.0\commons-io-2.15.0.jar;C:\Users\HO006043\.m2\repository\com\zaxxer\SparseBitSet\1.3\SparseBitSet-1.3.jar;C:\Users\HO006043\.m2\repository\org\apache\logging\log4j\log4j-api\2.21.1\log4j-api-2.21.1.jar;C:\Users\HO006043\.m2\repository\org\apache\poi\poi-ooxml\5.2.5\poi-ooxml-5.2.5.jar;C:\Users\HO006043\.m2\repository\org\apache\poi\poi-ooxml-lite\5.2.5\poi-ooxml-lite-5.2.5.jar;C:\Users\HO006043\.m2\repository\org\apache\xmlbeans\xmlbeans\5.2.0\xmlbeans-5.2.0.jar;C:\Users\HO006043\.m2\repository\org\apache\commons\commons-compress\1.25.0\commons-compress-1.25.0.jar;C:\Users\HO006043\.m2\repository\com\github\virtuald\curvesapi\1.08\curvesapi-1.08.jar;C:\Users\HO006043\.m2\repository\org\apache\logging\log4j\log4j-core\2.21.1\log4j-core-2.21.1.jar)
	at org.hid4java.HidDeviceManager.<init>(HidDeviceManager.java:87)
	at org.hid4java.HidServices.<init>(HidServices.java:81)
	at org.hid4java.HidManager.getHidServices(HidManager.java:72)
	at usbhid.comms.AppView.<init>(AppView.java:116)
	at usbhid.comms.Runner.main(Runner.java:10)
Caused by: java.lang.UnsatisfiedLinkError: Unable to load library 'hidapi':
The specified module could not be found.

The specified module could not be found.

The specified module could not be found.

Native library (win32-x86-64/hidapi.dll) not found in resource path (D:\COG_Java_USBHID_Developer\usbhid.comms\target\classes;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\commons-codec-1.16.0.jar;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\commons-collections4-4.4.jar;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\commons-io-2.15.0.jar;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\commons-math3-3.6.1.jar;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\poi-5.2.5.jar;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\poi-ooxml-5.2.5.jar;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\hid4java-0.8.0.jar;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\log4j-core-2.21.1.jar;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\xmlbeans-5.2.0.jar;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\log4j-api-2.21.1.jar;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\swt.jar;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\jna-5.13.0.jar;D:\COG_Java_USBHID_Developer\usbhid.comms\libs\jna-platform-5.13.0.jar;C:\Users\HO006043\.m2\repository\org\apache\poi\poi\5.2.5\poi-5.2.5.jar;C:\Users\HO006043\.m2\repository\commons-codec\commons-codec\1.16.0\commons-codec-1.16.0.jar;C:\Users\HO006043\.m2\repository\org\apache\commons\commons-collections4\4.4\commons-collections4-4.4.jar;C:\Users\HO006043\.m2\repository\org\apache\commons\commons-math3\3.6.1\commons-math3-3.6.1.jar;C:\Users\HO006043\.m2\repository\commons-io\commons-io\2.15.0\commons-io-2.15.0.jar;C:\Users\HO006043\.m2\repository\com\zaxxer\SparseBitSet\1.3\SparseBitSet-1.3.jar;C:\Users\HO006043\.m2\repository\org\apache\logging\log4j\log4j-api\2.21.1\log4j-api-2.21.1.jar;C:\Users\HO006043\.m2\repository\org\apache\poi\poi-ooxml\5.2.5\poi-ooxml-5.2.5.jar;C:\Users\HO006043\.m2\repository\org\apache\poi\poi-ooxml-lite\5.2.5\poi-ooxml-lite-5.2.5.jar;C:\Users\HO006043\.m2\repository\org\apache\xmlbeans\xmlbeans\5.2.0\xmlbeans-5.2.0.jar;C:\Users\HO006043\.m2\repository\org\apache\commons\commons-compress\1.25.0\commons-compress-1.25.0.jar;C:\Users\HO006043\.m2\repository\com\github\virtuald\curvesapi\1.08\curvesapi-1.08.jar;C:\Users\HO006043\.m2\repository\org\apache\logging\log4j\log4j-core\2.21.1\log4j-core-2.21.1.jar)
	at com.sun.jna.NativeLibrary.loadLibrary(NativeLibrary.java:323)
	at com.sun.jna.NativeLibrary.getInstance(NativeLibrary.java:483)
	at com.sun.jna.Library$Handler.<init>(Library.java:197)
	at com.sun.jna.Native.load(Native.java:622)
	at com.sun.jna.Native.load(Native.java:596)
[... cut ...] epository\org\apache\xmlbeans\xmlbeans\5.2.0\xmlbeans-5.2.0.jar;C:\Users\HO006043\.m2\repository\org\apache\commons\commons-compress\1.25.0\commons-compress-1.25.0.jar;C:\Users\HO006043\.m2\repository\com\github\virtuald\curvesapi\1.08\curvesapi-1.08.jar;C:\Users\HO006043\.m2\repository\org\apache\logging\log4j\log4j-core\2.21.1\log4j-core-2.21.1.jar)
		at com.sun.jna.Native.extractFromResourcePath(Native.java:1145)
		at com.sun.jna.NativeLibrary.loadLibrary(NativeLibrary.java:295)
		... 11 more

gary-rowe, 2024-03-29:
Looking over your output, it appears that there is a reference to `hid4java-0.8.0.jar` when the current latest release is `0.7.0` - could be a typo.

In general `hid4java` is referenced by downstream projects in Maven as follows:

```xml
<dependencies>

  <!-- hid4java for cross-platform HID USB -->
  <dependency>
    <groupId>org.hid4java</groupId>
    <artifactId>hid4java</artifactId>
    <version>0.7.0</version>
  </dependency>

</dependencies>
```

frankmeho, 2024-04-01:
Hi Gary,

The hid4java-0.8.0.jar was named by myself because I downloaded your
hid4java-develop and export it to a jar. Therefore I used another version
to distinguish it from the original hid4java-0.7.0.jar.

Sincerely,
Frank

gary-rowe, 2024-04-15:
Hi @frankmeho. I've release the `0.8.0` version of the library. Does this resolved the issue? Also, can you provide the exact platform id (e.g. `linex-aarch64` that is causing the issue?

frankmeho, 2024-04-17:
Hi Gary,

Thanks for your email.
I checked this IC platform and it’s Cortex-M4. The following web site is
the related information, fyi.
https://www.airoha.com/products/p/HanExuKmg0K35S1w

Sincerely,
Frank

frankmeho, 2024-04-17:
Hi Gary,

By the way, I’ve used hid4java-0.8.0 and I still encountered the same issue.

Thanks.

frankmeho, 2024-04-25:
Hi Gary,

I also provide a screenshot for Set Report/Get Report of this specific IC,
fyi.

Sincerely,
Frank

frankmeho, 2024-04-25:
So, its 0x06 is the Report ID for Set Report as well as the 0x07 the Report
ID for Get Report. We can’t get anything once we send Get Report.

closed 2026-11-11 by gary-rowe