gary-rowe/hid4java #146: How can i read the scanner data.

dearming623, 2024-03-30:
I have confirmed that VENDOR_ID and PRODUCT_ID are correct.

Found problem in hidDataReceived did not trigger. The reason is that the read function has not read data.

Please look at the problem. Thank you all. 

The operating environment is in the window system.
hidapi is version 0.14.0
The device is a scanning gun for HIM

**Here's my code**

public class BarcodeScannerReader {

    private static final int VENDOR_ID = 0x1eab;
    private static final int PRODUCT_ID = 0x1a03;

    public static void main(String[] args) throws InterruptedException {
        HidServices hidServices = HidManager.getHidServices();
        hidServices.addHidServicesListener(new HidServicesListener() {
            @Override
            public void hidDeviceAttached(HidServicesEvent event) {
                System.out.println("Device attached: " + event.getHidDevice());
            }

            @Override
            public void hidDeviceDetached(HidServicesEvent event) {
                System.out.println("Device detached: " + event.getHidDevice());
            }

            @Override
            public void hidFailure(HidServicesEvent event) {
                System.out.println("HID failure: " + event);
            }

            @Override
            public void hidDataReceived(HidServicesEvent event) {
                System.out.println("HID hidDataReceived: " + event);
            }
        });

        hidServices.start();
        HidDevice hidDevice = hidServices.getHidDevice(VENDOR_ID, PRODUCT_ID, null);

        if (hidDevice != null) {
            System.out.println("Found device: " + hidDevice);
            byte[] data = new byte[10];
            while (true) {
                int bytesRead = hidDevice.read(data, 1000);
                if (bytesRead > 0) {
                    String barcode = new String(data, 0, bytesRead);
                    System.out.println("Barcode scanned: " + barcode);
                }
            }
        } else {
            System.out.println("Device not found!");
        }

        hidServices.shutdown();
    }
}

dearming623, 2024-04-02:
I have checked out the latest source code and added a test class to the example.

dearming623, 2024-04-02:
> Can you verify if you're using the latest `develop-SNAPSHOT` release please?

I have checked out the latest source code and added a test class to the example

dearming623, 2024-04-02:
> Can you verify if you're using the latest `develop-SNAPSHOT` release please?

I have checked out the latest source code and added a test class to the example.

gary-rowe, 2024-04-15:
Thanks for the additional information. Does your device need any kind of initialisation to get it started? You might find the `Fido2AuthenticationExample` useful if you haven't already researched it.

closed 2026-11-11 by gary-rowe