gary-rowe/hid4java #154: WriteFile: (0x00000005) Access is denied

pooyamotorhead, 2024-06-20:
Hi
 I use Wireshark for reverse engineering the mice's application codes . I tried to send a set report request to the mice by using your lib . 
I faced a problem after sending Write request to the device  , it returns this Error : 
WriteFile: (0x00000005) Access is denied.

Here's My Code : 
```
    HidServices hidServices = HidManager.getHidServices();
    HidDevice device = hidServices.getHidDevice(20051, 21510, null);
    byte[] message = {0x21, 0x09, 0x07, 0x03, 0x01, 0x00 ,0x08, 0x00}; 
    byte reportid=0x07;
    int val = device.write(message, 8,reportid);
    if (val != -1) {
    System.out.println("> [" + val + "]");
    } else {
    System.err.println(device.getLastErrorMessage());
    }
```

Any hint is greatly appreciated

gary-rowe, 2024-06-24:
Can I verify that you've take a look at the [Troubleshooting article](https://github.com/gary-rowe/hid4java/wiki/Troubleshooting#my-device-doesnt-work) about device access with HID?

pooyamotorhead, 2024-06-28:
> Can I verify that you've take a look at the [Troubleshooting article](https://github.com/gary-rowe/hid4java/wiki/Troubleshooting#my-device-doesnt-work) about device access with HID?

Yes I checked that article 
I can open and read some fields like Usage page and Usage from the device 
But i cant send set report request to the device

pooyamotorhead, 2024-07-16:
Isn't there any way to solve this problem ?

gary-rowe, 2024-07-17:
Unfortunately not. My understanding is that it's part of the HID standard that keyboards and mice are special devices that are locked out. You'll need to use a lower level library.

pooyamotorhead, 2024-07-17:
Thank you
You mean USB lib , right ?

gary-rowe, 2024-07-17:
Yes, you should be using https://github.com/libusb/libusb for mouse and keyboard access.

closed 2024-07-17 by gary-rowe