gary-rowe/hid4java #158: Any idea why I don't get back info from my ledger wallet? Python works fine

fancellu, 2024-10-10:
**Describe the problem**

Using a little Java code, I don't get back what I expect to see from my ledger blue. Python works fine.

**Platform**

Knowing the platform greatly narrows down the potential causes of the problem.

- Platform Win11-x64:
- hid4java version0.8.0

**To Reproduce**

Steps to reproduce the behavior:

Run below JavaHidExample code
```
package ledge;

import org.hid4java.*;

public class JavaHidExample {
    public static void main(String[] args) throws InterruptedException {
        HidServicesSpecification hidSpec = new HidServicesSpecification();

        HidServices hidServices = HidManager.getHidServices(hidSpec);
        hidServices.start();

        hidServices.getAttachedHidDevices().forEach(device -> {

            if ("Ledger".equals(device.getManufacturer())) {
                System.out.println("  ledger device!!!!!!!!!!!!!!!");

                System.out.println(device);
                boolean opened = device.open();
                System.out.println(opened);

                byte[] byteArray = {(byte)0xE0, 0x01, 0x00, 0x00};

                int bytesWritten = device.write(byteArray,byteArray.length, (byte) 0);
                System.out.println(bytesWritten);

                byte[] bytesRead = device.readAll(1000);
                System.out.println(bytesRead.length);
                for (byte b : bytesRead) {
                    System.out.print(b);
                }
            }
        });
    }
}
```

Output is

```
  ledger device!!!!!!!!!!!!!!!
HidDevice [path=\\?\HID#VID_2C97&PID_0000&MI_00#a&2efb5d03&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}, vendorId=0x2c97, productId=0x0, serialNumber=0001, releaseNumber=0x200, manufacturer=Ledger, product=Blue, usagePage=0xffffffa0, usage=0x1, interfaceNumber=0]
true
65
64
-32100000000000000000000000000000000000000000000000000000000000000
```

**Expected behavior**

Running v similar python code works fine
```
from ledgercomm import Transport

transport = Transport(interface="hid", debug=True)

transport.send(cla=0xe0, ins=0x01, p1=0, p2=0)

sw, response = transport.recv()  # type: int, bytes

print(sw)
print(response)
```

```
ledgercomm - DEBUG - hid.enumerate(), devices: [{'path': b'\\\\?\\HID#VID_2C97&PID_0000&MI_00#a&2efb5d03&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}', 'vendor_id': 11415, 'product_id': 0, 'serial_number': '0001', 'release_number': 512, 'manufacturer_string': 'Ledger', 'product_string': 'Blue', 'usage_page': 65440, 'usage': 1, 'interface_number': 0, 'bus_type': 1}]
ledgercomm - DEBUG - => e001000000
36864
b'1\x01\x00\x04\x052.1.1\x04\xa6\x00\x00\x00\t3.4-hw15\x00 J\x84j\xbfVe\xf7\xe2s\xf4\xcf\x9f\xdd\xe8f\xcd\xdc3J\xde\x1d\xd4\x85Y\xaa&6\r>\x1fR\xa1'
ledgercomm - DEBUG - <= 3101000405322e312e3104a600000009332e342d6877313500204a846abf5665f7e273f4cf9fdde866cddc334ade1dd48559aa26360d3e1f52a1 9000

```

No doubt I'm doing something silly, but I have no clue, and there don't seem to be very much example code out there

Thanks

gary-rowe, 2024-10-11:
I'm no expert on Ledger devices, but it could be that the Python library is adjusting the values under the covers, perhaps presenting them as `int` instead of `byte`. I've found in the past that it is generally better to use the manufacturer's libraries, e.g. LedgerCOMM, rather than roll a custom solution.

fancellu, 2024-10-11:
Well, I want to do this from Java, hence the desire to use this library, and not a Python solution. The docs around this space are so vague, chopped up, out of date etc. 

I'd just like to know EXACTLY what I should be sending. Its astonishing how badly documented things are in the crypto space.

fancellu, 2024-10-11:
Ok, making progress by going through Python libs. And yes, they do lots of undocumented bit twiddling, sigh. 

Anyway, consider this ticket closed. hid4java works fine.

closed 2024-10-11 by fancellu