gary-rowe/hid4java #161: On Ubuntu 24.04.1 LTS examples run under Maven do not terminate cleanly

tim-littlefair, 2025-02-17:
**Describe the problem**

In the hid4java wiki, there are recommendations to run the test/example programs using Maven commands of the form:

`mvn clean test exec:java -Dexec.classpathScope="test" -Dexec.mainClass="org.hid4java.examples.UsbHidEnumerationExample"`

I am running on an Intel machine with Ubuntu 24.04.1 LTS installed, when run any of the examples I find that the shutdown is delayed, and only completes after warnings and exceptions are logged (see log messages below).

You may find that this issue has arisen before, so here is a checklist of useful links to provide assistance and potentially avoid duplicate issues:
- [X] I have read the [Troubleshooting Guide](https://github.com/gary-rowe/hid4java/wiki/Troubleshooting) and this is not covered
- [X] I have done a [quick search of previous issues](https://github.com/gary-rowe/hid4java/issues?q=is%3Aissue+hidraw) and this needs addressing

Please provide a clear and concise description of what the problem is below:

**Platform**

Knowing the platform greatly narrows down the potential causes of the problem.

- Platform: `linux-x86-64`
- OS version `Ubuntu 22.04.1`
- `hid4java` versions: `0.8.0`, `develop`

**To Reproduce**

Steps to reproduce the behavior:

`sudo mvn clean test exec:java -Dexec.classpathScope="test" -Dexec.mainClass="org.hid4java.examples.UsbHidEnumerationExample"`
NB sudo is required on my environment because I am choosing not to put udev rules in place before the application I am working on has reached the stage of needing an installer.

**Expected behavior**

On completion of the enumerated device report, the following two lines should appear, 30 seconds apart.

```
Waiting 30s to demonstrate attach/detach handling. Watch for slow response after write if configured.
Triggering shutdown...
```
The program should exit immediately after the second line appears.

**Actual behaviour**
The following two lines appear:
```
Waiting 30s to demonstrate attach/detach handling. Watch for slow response after write if configured.
Triggering shutdown...
```
but the program does not exit for some time (probably 15 seconds, haven't measured it).
Before the program does exit it prints the following lines:
```
[WARNING] thread Thread[hid4java event worker,5,org.hid4java.examples.UsbHidEnumerationExample] was interrupted but is still alive after waiting at least 15000msecs
[WARNING] thread Thread[hid4java event worker,5,org.hid4java.examples.UsbHidEnumerationExample] will linger despite being asked to die via interruption
[WARNING] thread Thread[hid4java event worker,5,org.hid4java.examples.UsbHidEnumerationExample] will linger despite being asked to die via interruption
[WARNING] thread Thread[hid4java event worker,5,org.hid4java.examples.UsbHidEnumerationExample] will linger despite being asked to die via interruption
[WARNING] NOTE: 3 thread(s) did not finish despite being asked to via interruption. This is not a problem with exec:java, it is a problem with the running code. Although not serious, it should be remedied.
[WARNING] Couldn't destroy threadgroup org.codehaus.mojo.exec.ExecJavaMojo$IsolatedThreadGroup[name=org.hid4java.examples.UsbHidEnumerationExample,maxpri=10]
java.lang.IllegalThreadStateException
    at java.lang.ThreadGroup.destroy (ThreadGroup.java:803)
    at org.codehaus.mojo.exec.ExecJavaMojo.execute (ExecJavaMojo.java:330)
    at org.apache.maven.plugin.DefaultBuildPluginManager.executeMojo (DefaultBuildPluginManager.java:137)
    at org.apache.maven.lifecycle.internal.MojoExecutor.doExecute2 (MojoExecutor.java:370)
    at org.apache.maven.lifecycle.internal.MojoExecutor.doExecute (MojoExecutor.java:351)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:215)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:171)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:163)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:117)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:81)
    at org.apache.maven.lifecycle.internal.builder.singlethreaded.SingleThreadedBuilder.build (SingleThreadedBuilder.java:56)
    at org.apache.maven.lifecycle.internal.LifecycleStarter.execute (LifecycleStarter.java:128)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:298)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:192)
    at org.apache.maven.DefaultMaven.execute (DefaultMaven.java:105)
    at org.apache.maven.cli.MavenCli.execute (MavenCli.java:960)
    at org.apache.maven.cli.MavenCli.doMain (MavenCli.java:293)
    at org.apache.maven.cli.MavenCli.main (MavenCli.java:196)
    at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0 (Native Method)
    at jdk.internal.reflect.NativeMethodAccessorImpl.invoke (NativeMethodAccessorImpl.java:77)
    at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke (DelegatingMethodAccessorImpl.java:43)
    at java.lang.reflect.Method.invoke (Method.java:569)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launchEnhanced (Launcher.java:283)
    at org.codehaus.plexus.classworlds.launcher.Launcher.launch (Launcher.java:226)
    at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode (Launcher.java:407)
    at org.codehaus.plexus.classworlds.launcher.Launcher.main (Launcher.java:348)
```

**Additional information**

I believe that this behaviour may be the result of a problem with Maven, described in this StackOverflow response:
https://stackoverflow.com/a/77783869

I have a local copy of a linux-specific shell script to invoke the tests without using mvn as a wrapper.  With this script the problem does not occur.
I am happy to do a pull request with this tomorrow if it would be welcome.

gary-rowe, 2025-02-18:
Hi Tim

Thanks for getting in touch. I’m AFK for this for a few days, but if you’d
like to set up the pull request I’m happy to look at it.

OBenz94, 2025-04-16:
Hello,
I had a problem in the beginning to get the information of the HID device which is a ship console in my case.
Then i have used the 0.8.0 instead of 0.7.0 in the pom and it works, however i got this message
Exception in thread "Thread-2" Exception in thread "Thread-3"
java.lang.IllegalStateException: Device has not been opened
	at org.hid4java.HidDevice.read(HidDevice.java:397)
	at org.hid4java.HidDevice.read(HidDevice.java:444)
	at de.jade.azimuthconsole.AzimuthControllerPoller.pollAndReadBytes(AzimuthControllerPoller.java:97)
	at de.jade.azimuthconsole.AzimuthControllerPoller.run(AzimuthControllerPoller.java:63)
	at java.base/java.lang.Thread.run(Thread.java:1583)
java.lang.IllegalStateException: Device has not been opened
	at org.hid4java.HidDevice.read(HidDevice.java:397)
	at org.hid4java.HidDevice.read(HidDevice.java:444)
	at de.jade.azimuthconsole.AzimuthControllerPoller.pollAndReadBytes(AzimuthControllerPoller.java:97)
	at de.jade.azimuthconsole.AzimuthControllerPoller.run(AzimuthControllerPoller.java:63)
	at java.base/java.lang.Thread.run(Thread.java:1583).
why the device is not open, even though i got the information and lsusb recognize my both HID device.

Thank you in advance.

closed 2026-11-11 by gary-rowe